BambuTools/bambulabs_api #38: cant connect

CCJv, 2024-07-27:
not able to connect to the p1p printer
ERROR:root:Not connected to the MQTT server
Printer status: UNKNOWN
ERROR:root:Wrong access code or IP

ive got the right access code, ip and s/n

cyx258, 2024-07-28:
I got the same problom,have you fix it?

ohmdelta, 2024-07-30:
Can you check if the printer is in lan mode - the api has only been tested in non-lan only mode. 
Can you also scan the open ports of your P1P printer - check that 8883, 990 and 6000 are all open.

CCJv, 2024-07-30:
so in a different script im able to connect and publish a request (turn led on)
`import paho.mqtt.client as mqtt
import json
import ssl

# Configuration
BROKER_HOST = '0.0.0.0'
BROKER_PORT = 8883
TOPIC_REQUEST = 'device/xxxxxx/request'
TOPIC_REPORT = 'device/xxxxxx/report'
USERNAME = 'bblp'
PASSWORD = 'xxxxxx'

# MQTT Callback Functions
def on_connect(client, userdata, flags, rc):
    print(f"Connected with result code {rc}")
    client.subscribe(TOPIC_REPORT)
    print(f"Subscribed to topic: {TOPIC_REPORT}")

def on_message(client, userdata, msg):
    print(f"Received message '{msg.payload.decode()}' on topic '{msg.topic}'")

def publish_request(client, request_data):
    payload = json.dumps(request_data)
    client.publish(TOPIC_REQUEST, payload)
    print(f"Published request: {payload}")

# MQTT Client Setup
client = mqtt.Client()

# Set username and password
client.username_pw_set(USERNAME, PASSWORD)

# Attach callback functions
client.on_connect = on_connect
client.on_message = on_message

# Configure TLS, disable certificate verification
client.tls_set(certfile=None, keyfile=None, cert_reqs=ssl.CERT_NONE, tls_version=ssl.PROTOCOL_TLS, ciphers=None)

# Connect to the broker
client.connect(BROKER_HOST, BROKER_PORT, 60)

# Publish a sample request
request_data = {
"system": {
                                    "sequence_id": "0", 
                                    "command": "ledctrl", 
                                    "led_node": "chamber_light", 
                                    "led_mode": "on",
                                    "led_on_time": 500, 
                                    "led_off_time": 500, 
                                    "loop_times": 0, 
                                    "interval_time": 0
                                }
}
publish_request(client, request_data)

# Loop forever
client.loop_forever()
`

Published request: {"system": {"sequence_id": "0", "command": "ledctrl", "led_node": "chamber_light", "led_mode": "on", "led_on_time": 500, "led_off_time": 500, "loop_times": 0, "interval_time": 0}}
Connected with result code 0
Subscribed to topic: device/01S00C342900092/report
Received message '{"print":{"wifi_signal":"-68dBm","lights_report":[{"node":"chamber_light","mode":"on"}],"command":"push_status","msg":1,"sequence_id":"1922"}}' on topic 'device/01S00C342900092/report'

ohmdelta, 2024-07-31:
Good to hear you got the script working. Would you be able to share the first script

CCJv, 2024-07-31:
sure thing. so I pulled the repo and simply ran the script examples\basic\basic.py after amending the correct IP, serial and access code. I then created a separate test script which uses the mqtt client directly. ive managed to get connect to the printer and send a request to turn the led light on and off.  I'm still trying to troubleshoot why the test.py script in the repo doesnt work? you mentioned that youve have this working on a p1p or x1c?

```
import bambulabs_api as bl

IP = 'xxxx'
SERIAL = 'xxxxx'
ACCESS_CODE = 'xxxxx'

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to Bambulabs 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, SERIAL, ACCESS_CODE)

    # Connect to the Bambulabs 3D printer
    printer.connect()

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()
```
Output
Connecting to Bambulabs 3D printer
IP: 192.168.0.86
Serial: 01S00C342900092
Access Code: 12410177
Starting camera thread.
ERROR:root:Not connected to the MQTT server
Printer status: UNKNOWN
ERROR:root:Wrong access code or IP

```
import paho.mqtt.client as mqtt
import json
import ssl

BROKER_HOST = 'xxxxx'
BROKER_PORT = 8883
TOPIC_REQUEST = 'device/xxxxx/request'
TOPIC_REPORT = 'device/xxxxx/report'
USERNAME = 'bblp'
PASSWORD = 'xxxxx'

def on_connect(client, userdata, flags, rc):
    print(f"Connected with result code {rc}")
    client.subscribe(TOPIC_REPORT)
    print(f"Subscribed to topic: {TOPIC_REPORT}")

def on_message(client, userdata, msg):
    print(f"Received message '{msg.payload.decode()}' on topic '{msg.topic}'")

def publish_request(client, request_data):
    payload = json.dumps(request_data)
    client.publish(TOPIC_REQUEST, payload)
    print(f"Published request: {payload}")

client = mqtt.Client()

client.username_pw_set(USERNAME, PASSWORD)

client.on_connect = on_connect
client.on_message = on_message

client.tls_set(certfile=None, keyfile=None, cert_reqs=ssl.CERT_NONE, tls_version=ssl.PROTOCOL_TLS, ciphers=None)

client.connect(BROKER_HOST, BROKER_PORT, 230)

request_data = {
                                "system": {
                                    "sequence_id": "0", 
                                    "command": "ledctrl", 
                                    "led_node": "chamber_light", 
                                    "led_mode": "off",
                                    "led_on_time": 500, 
                                    "led_off_time": 500, 
                                    "loop_times": 0, 
                                    "interval_time": 0
                                }
                            
}
publish_request(client, request_data)

client.loop_forever()
```

output

Published request: {"system": {"sequence_id": "0", "command": "ledctrl", "led_node": "chamber_light", "led_mode": "off", "led_on_time": 500, "led_off_time": 500, "loop_times": 0, "interval_time": 0}}
Connected with result code 0

ohmdelta, 2024-07-31:
Hiya, there's a small typo in the example code: the access code and serial are the wrong way round 
Line 16 should be:
```python
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)
```

closed 2024-08-01 by acse-ci223