BambuTools/bambulabs_api #41: Can't connect with the simple example

yanshay, 2024-08-08:
I'm unable to connect with the example (code as of today, after the fix made in other closed issue that reported connection issues). 
I have the correct IP/Serial/Access Code, I'm not on lan-only mode and have other software (OctoEverywhere) that connects to the printer properly from the same local network.

I get the following log:
```
Starting bambulabs_api example
Connecting to Bambulabs 3D printer
IP: 192.168.10.78
Serial: 01P00A3A2900822
Access Code: 26259876
Starting camera thread.
ERROR:root:Not connected to the MQTT server
Printer status: UNKNOWN
```
and then it hangs

if I press `^C`, I get:

```
Traceback (most recent call last):
  File "/Users/my_user/ProgProj/bambulab/try1.py", line 27, in <module>
    printer.disconnect()
  File "/usr/local/lib/python3.11/site-packages/bambulabs_api/client.py", line 47, in disconnect
    self.__printerCamera.stop()
  File "/usr/local/lib/python3.11/site-packages/bambulabs_api/camera_client.py", line 31, in stop
    self.__thread.join()
  File "/usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/threading.py", line 1119, in join
    self._wait_for_tstate_lock()
  File "/usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/threading.py", line 1139, in _wait_for_tstate_lock
    if lock.acquire(block, timeout):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

ohmdelta, 2024-08-12:
There's a small issue with previous versions of the example. Try using the example in the README

yanshay, 2024-08-12:
> There's a small issue with previous versions of the example. Try using the example in the README

I'm using the code from the README.
Maybe something changed with recent firmware updates?

Here is the code I'm using (with a Bambulab P1S):
   ```
    print('Starting bambulabs_api example')
    print('Connecting to Bambulabs 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the Bambulabs 3D printer
    printer.connect()

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()
```

ohmdelta, 2024-08-12:
The example should be working, could you check again that the configs are all correct?

yanshay, 2024-08-12:
The configs (IP, ACCESS_CODE, SERIAL) are surely correct because I use the same variables (copy/pasted) in the 2nd code provided in https://github.com/acse-ci223/bambulabs_api/issues/38#issuecomment-2261420451 , and it is connecting and working.

I'm not missing any other config there, right?

ohmdelta, 2024-08-12:
Are you using the pip package or have you cloned the repo? If you're using pip it could be a version issue.

yanshay, 2024-08-12:
I'm using pip, when I check the version, this is what I see (which seems newer than the latest release published on this repo (2.3.1) (but not uncommon to see versions released without the release showing on github):

```
Name: bambulabs_api
Version: 2.3.3
Summary: API for BambuLabs 3D Printers over MQTT
Home-page: https://github.com/acse-ci223/bambulabs_api
Author: Chris Ioannidis
Author-email: Chris Ioannidis <chris.ioannidis23@imperial.ac.uk>
License:
Location: /usr/local/lib/python3.11/site-packages
Requires: paho-mqtt
Required-by:
```

Could the released version be not working and need require using the clone?

ohmdelta, 2024-08-12:
The released version should be working.  2.3.1 has been tested

yanshay, 2024-08-12:
So 2.3.3 I have should work?

ohmdelta, 2024-08-12:
Yes, it should be working

CCJv, 2024-08-12:
@yanshay I get the same issue. unable to connect.

ohmdelta, 2024-08-12:
The issue seems to be that the pushall command (which triggers the printer to publish a more complete report) is temperamental

CCJv, 2024-08-21:
thanks, fixed by putting a sleep in the  def __publish_command function within the PrinterMQTTClient class

JustJace1, 2024-11-17:
im getting this same issue how do i fix this?

mchrisgm, 2024-11-17:
@JustJace1 Have you updated to the latest version and using the example found in the README or the examples folder?

JustJace1, 2024-11-17:
yes it still wont connect

ohmdelta, 2024-11-24:
This should be fixed now in #64

closed 2024-11-25 by mchrisgm