BambuTools/bambulabs_api #48: Compatibility with A1?

NelsonLive, 2024-10-23:
Trying to connect to my bambu lab A1 using the example code on the pip page and getting this error:

Starting camera thread.
ERROR:root:Not connected to the MQTT server
Printer status: UNKNOWN

acse-ci223, 2024-10-23:
We currently do not support A1s because we don't have any machines to test on. Will update this once we get the API working with A1 and A1 mini.

In the meantime, please fill free to debug if you have the technical knowledge.
Thanks :)

vaaski, 2024-11-05:
I have a A1 and I'm referencing this repo to implement the API in TypeScript. So far I didn't run into any issues on my part but I'll report back with any differences I encounter.

https://github.com/vaaski/bambu.ts
https://github.com/vaaski/bambu.ts/blob/main/src/index.ts
https://github.com/vaaski/bambu.ts/blob/main/example.ts

mchrisgm, 2024-11-05:
@vaaski Silly question but did you try `--network=host` like shown in the docker documentation?

vaaski, 2024-11-05:
> @vaaski Silly question but did you try `--network=host` like shown in the docker documentation?

Yeah, if you scroll through the command I included to run the container you can see it there :)
Could still be a docker issue, I'll try to debug it on friday when I'm back home. I also was at work and VPNing to home, but my TypeScript version works

NelsonLive, 2024-11-12:
If you send me the code I can test it on my A1. @mchrisgm

tekierz, 2024-11-17:
Yeah, I'm also getting the same errors as @NelsonLive 

certain things work like I can control the LED, but all the states and AMS seem to be just reporting unknown:

```2024-11-17 02:21:52,654 - Printer Status: {'state': 'UNKNOWN', 'print_status': 'UNKNOWN', 'remaining_time': None, 'completion': None, 'print_speed': 100, 'bed_temp': 0.0, 'nozzle_temp': 0.0, 'current_file': '', 'light_state': 'unknown', 'skipped_objects': []}
2024-11-17 02:21:52,654 - ----- Printer Status Update -----
2024-11-17 02:21:52,655 - state: UNKNOWN
2024-11-17 02:21:52,655 - print_status: UNKNOWN
2024-11-17 02:21:52,655 - remaining_time: None
2024-11-17 02:21:52,655 - completion: None
2024-11-17 02:21:52,655 - print_speed: 100
2024-11-17 02:21:52,655 - bed_temp: 0.0
2024-11-17 02:21:52,655 - nozzle_temp: 0.0
2024-11-17 02:21:52,655 - current_file: 
2024-11-17 02:21:52,655 - light_state: unknown
2024-11-17 02:21:52,655 - skipped_objects: []
2024-11-17 02:21:52,655 - ------------------------------
2024-11-17 02:21:52,655 - Current GCode state: UNKNOWN
2024-11-17 02:21:52,655 - Current Print status: UNKNOWN
2024-11-17 02:21:52,655 - Printer is ready```

mchrisgm, 2024-11-17:
@tekierz Are you also trying to connect to an A1?

I'll be adding a debugging feature so we can see what's happening with the A1 printers.

tekierz, 2024-11-17:
@mchrisgm yeah, I'm also trying to connect to an A1

NelsonLive, 2024-11-17:
Tell me when you add the debugging feature so I can try again!

ohmdelta, 2024-11-28:
Can you try with the latest code

NelVilella, 2024-11-29:
Yes, I will try the example code again with the A1. Do I need to update the pip or something?

ohmdelta, 2024-11-29:
You can run `pip install bambulabs_api==2.5.0`

NelsonLive, 2024-11-30:
[image]
Status_ UNKNOWN

ohmdelta, 2024-11-30:
could you print the mqtt dump - `printer.mqtt_dump()`, might need to wait a couple seconds

NelsonLive, 2024-12-01:
MQTT Dump: {'bed_temper': 13.53125, 'command': 'push_status', 'msg': 1, 'sequence_id': '44254'}
MQTT Dump: {'bed_temper': 13.53125, 'command': 'push_status', 'msg': 1, 'sequence_id': '44257', 'upgrade_state': {'sequence_id': 0, 'progress': 
'', 'status': '', 'consistency_request': False, 'dis_state': 1, 'err_code': 0, 'force_upgrade': False, 'message': '0%, 0B/s', 'module': '', 'new_version_state': 1, 'cur_state_code': 0, 'new_ver_list': [{'name': 'ota', 'cur_ver': '01.03.00.00', 'new_ver': '01.03.01.02', 'cur_release_type': 3, 'new_release_type': 3}]}, 'ipcam': {'ipcam_dev': '1', 'ipcam_record': 'enable', 'timelapse': 'disable', 'resolution': '1080p', 'tutk_server': 'disable', 'mode_bits': 3}, 'xcam': {'buildplate_marker_detector': True}, 'upload': {'status': 'idle', 'progress': 0, 'message': ''}, 'nozzle_temper': 12.90625, 'nozzle_target_temper': 0, 'bed_target_temper': 0, 'chamber_temper': 5, 'mc_print_stage': '1', 'heatbreak_fan_speed': '0', 'cooling_fan_speed': '0', 'big_fan1_speed': '0', 'big_fan2_speed': '0', 'mc_percent': 100, 'mc_remaining_time': 0, 'ams_status': 0, 'ams_rfid_status': 0, 'hw_switch_state': 1, 'spd_mag': 100, 'spd_lvl': 2, 'print_error': 0, 'lifecycle': 'product', 'wifi_signal': '-74dBm', 'gcode_state': 'FINISH', 'gcode_file_prepare_percent': '100', 'queue_number': 0, 'queue_total': 0, 'queue_est': 0, 'queue_sts': 0, 'project_id': '175828230', 'profile_id': '167030868', 'task_id': '352186234', 'subtask_id': '352186235', 'subtask_name': 'output (41).stl', 'gcode_file': '', 'stg': [], 'stg_cur': 255, 'print_type': 'idle', 'home_flag': 847136152, 'mc_print_line_number': '44340', 'mc_print_sub_stage': 0, 'sdcard': True, 'force_upgrade': False, 'mess_production_state': 'active', 'layer_num': 57, 'total_layer_num': 57, 's_obj': [], 'filam_bak': [], 'fan_gear': 0, 'nozzle_diameter': '0.4', 'nozzle_type': 'stainless_steel', 'cali_version': 0, 'hms': [], 'online': {'ahb': False, 'rfid': False, 'version': 1201259944}, 'ams': {'ams': [], 'ams_exist_bits': '0', 'tray_exist_bits': '0', 'tray_is_bbl_bits': '0', 'tray_tar': '255', 'tray_now': '254', 'tray_pre': '254', 'tray_read_done_bits': '0', 'tray_reading_bits': '0', 'version': 2, 'insert_flag': True, 'power_on_flag': False}, 'vt_tray': {'id': '254', 'tag_uid': '0000000000000000', 'tray_id_name': '', 'tray_info_idx': '', 'tray_type': '', 'tray_sub_brands': '', 'tray_color': 'FFFFFFFF', 'tray_weight': '0', 'tray_diameter': '0.00', 'tray_temp': '0', 'tray_time': '0', 'bed_temp_type': '0', 'bed_temp': '0', 'nozzle_temp_max': '0', 'nozzle_temp_min': '0', 'xcam_info': '000000000000000000000000', 'tray_uuid': '00000000000000000000000000000000', 'remain': 0, 'k': 0.019999999552965164, 'n': 1, 'cali_idx': -1}, 'lights_report': [{'node': 'chamber_light', 'mode': 'off'}]}

ohmdelta, 2024-12-01:
Thanks @NelsonLive, looks like this is working for a1. Could you check the exposed apis?

NelsonLive, 2024-12-01:
How can I check that?

ohmdelta, 2024-12-01:
Once you get the dump, run get state, get bed temp etc. These should be exposed by the printer client in the example code

chickenorbeef, 2024-12-03:
i have an A1 mini, just ran the example code after `pip install bambulabs_api==2.5.0`, and am not getting any MQTT messages. will keep tinkering to see if i have any luck

ohmdelta, 2024-12-03:
> i have an A1 mini, just ran the example code after `pip install bambulabs_api==2.5.0`, and am not getting any MQTT messages. will keep tinkering to see if i have any luck

@chickenorbeef would you  be able to share the output that you're seeing. I've also created a couple more enhanced examples in the `examples` feature branch - <https://github.com/mchrisgm/bambulabs_api/blob/examples/examples/Basic/basic_subscription.py>

thenewguy, 2024-12-29:
@mchrisgm thinking about getting an a1 mini - do you have any reference tests in another repository you are using to verify interoperability? I didn't see any tests in the repo to reference for implementation

shooter-z, 2025-01-06:
Result of running https://github.com/mchrisgm/bambulabs_api/blob/main/examples/Basic/basic_subscription.py on A1 with AMS:
[basic.txt](https://github.com/user-attachments/files/18323672/basic.txt)

Let me know if you need anything else tested.

Also, api freezes completely when terminating script. Based on my tests it stops at printer.disconnect() And by completely I mean it, ctrl+c does not cancel it, only way is to close the cmd window.

Python 3.10.11
bambulabs_api 2.5.5
paho-mqtt 2.1.0
pillow 11.0.0

ohmdelta, 2025-01-07:
> Result of running https://github.com/mchrisgm/bambulabs_api/blob/main/examples/Basic/basic_subscription.py on A1 with AMS: [basic.txt](https://github.com/user-attachments/files/18323672/basic.txt) [...]

Thanks, closing issue as A1 confirmed to be supported

closed 2025-01-07 by ohmdelta