BambuTools/bambulabs_api #71: No MQTT Messages while connection works

Fireobirdy, 2024-12-06:
Hi,

i tried the example script to connect to my P1S. The connection seems to work. The light is turning off and on again. So commands can be sent to the printer. But i cannot read any information from the printer. 

`Starting bambulabs_api example
Connecting to BambuLab 3D printer
IP: 192.168.178.71
Serial: 01P00A431300XXX
Access Code: XXXXXXXX
Starting camera thread.
Printer status: UNKNOWN
`

If i try to read back the nozzle temperature i get 0, even if i set it to any other temp. It looks like, that no mqtt message is comming from the printer. Do i have to send a command? I tried both LAN modes and restarted several times. Firmware is latest non-beta. Have i overseen something?

ohmdelta, 2024-12-06:
What version of bambulabs api are you using. You may also need to wait 30 seconds before getting the first mqtt messages

WaskaLabs, 2024-12-16:
I assume the OP got this figured out, however I am having the same issue I think.

```
...
Starting camera thread.
Printer status: UNKNOWN
ERROR:root:Error occurred: [WinError 10061] No connection could be made because the target machine actively refused it
ERROR:root:Error occurred: [WinError 10061] No connection could be made because the target machine actively refused it
...
```

The light appears to be turning on, but the error keeps printing (past 30 seconds, I waited a few minutes to see if it changed). Using the example code provided.

API Version: 2.5.2
PY Version 3.1.3

If this is a stupid network issue of some sort please forgive, embedded guy here out of my element.

Thanks for any help!

WaskaLabs, 2024-12-17:
- Or this error

`A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond`

ohmdelta, 2024-12-19:
> I assume the OP got this figured out, however I am having the same issue I think.
>
> ```
> ...
> Starting camera thread.
> Printer status: UNKNOWN
> ERROR:root:Error occurred: [WinError 10061] No connection [...]

We haven't properly tested against the A1 or X1 printers yet - only P1 so far but my suspicion is that this is probably a network/config issue (possibly incorrect params being given to the api). Is the IP you've given to the api the same as the one being displayed on your printer (could you also check the open ports on your printer - you can run `nmap -sn <YOUR PRINTER IP> `  in Linux - 8883, 6000 and 990 should be available. If not there might be a router issue?). Also check that your access code and device id match the printer's. GL :)

WaskaLabs, 2024-12-20:
Thanks - This is a P1S, got it working. My issue was with python not the lib. Appreciate the response!

JustJace1, 2024-12-30:
same issue i ran  a nmap scan i only have 990 and port 6000 open is that ok im on a p1p

ohmdelta, 2024-12-30:
> same issue i ran  a nmap scan i only have 990 and port 6000 open is that ok im on a p1p
> 

That's strange, are you running the printer in lan mode (this lib has been tested with lan mode off)? You definitely need to have port 8883 open to be able to use this api, might be a network or printer issue. Could you check if you can access your printer via the orcaslicer/bambu studio or bambu handy?

JustJace1, 2025-01-01:
im not on lan made i can use orcaslicer just fine idk why it wont work

ohmdelta, 2025-01-02:
> im not on lan made i can use orcaslicer just fine idk why it wont work

Could you try running the basic connection example: <https://github.com/mchrisgm/bambulabs_api/blob/v2.5.2/examples/Basic/basic_subscription.py>? This should work on firmware version 01.07.00.00 (and has been tested against 01.05.*) in non-lan mode as long as you're not blocking any ports

ohmdelta, 2025-01-24:
Closing - Stale

closed 2025-01-24 by ohmdelta