BambuTools/bambulabs_api #78: Question: Incorrectly named parameter in start_print_3mf

suchmememanyskill, 2024-12-21:
Referring to https://github.com/mchrisgm/bambulabs_api/blob/main/bambulabs_api/mqtt_client.py#L292

When i intercept messages between bambu studio and my A1 printer, i see the following:
"vibration_cali": false,

But in this library i see: 
"vibration_calibration": True,

Just double checking, is that correct?

ohmdelta, 2025-01-02:
Think you're correct - might have changed between firmware updates, pr has been opened for this with example print gcode example

suchmememanyskill, 2025-01-03:
For reference, my a1 mini is running firmware 01.02.00.00

ohmdelta, 2025-01-03:
> For reference, my a1 mini is running firmware 01.02.00.00

I don't have an a1 so can't test this unfortunately. Could you try the print geode example using the opened pr?

closed 2025-01-05 by mchrisgm