BambuTools/bambulabs_api #85: AMS Filament trays not populated

AlanPeters, 2025-01-08:
Working on hooking up this up to my X1 Carbon and am having issues with the filament trays in the AMS. It looks like AMS support is pretty recent so I'm not sure if this is a bug or not implemented yet. 

Running using the development directions @
 f7ec5ea (tag: 2.5.5, origin/main, origin/HEAD) Merge pull request #81 from mchrisgm/v2.5.5

Test code: 
```
    dump = printer.mqtt_dump()

    ams_hub = printer.ams_hub()
    print("AMS:")
    print(vars(ams_hub[0]))

    print("printer.mgtt_dump()[\"AMS\"]")
    print(pprint.pprint(dump["ams"]))
```
Output
```
printer.ams_hub[0] :
{'filament_trays': {}, 'humidity': 5, 'temperature': 34.9}
printer.mgtt_dump()["AMS"]
{'ams': [{'humidity': '5',
          'id': '0',
          'temp': '34.9',
          'tray': [{'bed_temp': '0',
                    'bed_temp_type': '0',
                    'cali_idx': -1,
                    'cols': ['9B9EA0FF'],
                    'ctype': 0,
                    'drying_temp': '55',
                    'drying_time': '8',
                    'id': '0',
                    'nozzle_temp_max': '230',
                    'nozzle_temp_min': '190',
                    'remain': 60,
                    'tag_uid': 'CA4C27FE00000100',
                    'tray_color': '9B9EA0FF',
                    'tray_diameter': '1.75',
                    'tray_id_name': 'A01-D3',
                    'tray_info_idx': 'GFA01',
                    'tray_sub_brands': 'PLA Matte',
                    'tray_type': 'PLA',
                    'tray_uuid': 'B2D4D17E1A5E4EDAAE938CF0279113AA',
                    'tray_weight': '1000',
                    'xcam_info': 'D007D007E803E8030000803F'},
                   {'bed_temp': '0',
                    'bed_temp_type': '0',
                    'cali_idx': -1,
                    'cols': ['BC0900FF'],
                    'ctype': 0,
                    'drying_temp': '65',
                    'drying_time': '8',
                    'id': '1',
                    'nozzle_temp_max': '260',
                    'nozzle_temp_min': '230',
                    'remain': 35,
                    'tag_uid': '3157FC0F00080100',
                    'tray_color': 'BC0900FF',
                    'tray_diameter': '1.75',
                    'tray_id_name': 'G02-R0',
                    'tray_info_idx': 'GFG02',
                    'tray_sub_brands': 'PETG HF',
                    'tray_type': 'PETG',
                    'tray_uuid': '61096C73F27041A4B3774D89C47A5A0C',
                    'tray_weight': '1000',
                    'xcam_info': '803E803EBC0220033333733F'},
                   {'bed_temp': '0',
                    'bed_temp_type': '0',
                    'cali_idx': -1,
                    'cols': ['FFFFFFFF'],
                    'ctype': 0,
                    'drying_temp': '55',
                    'drying_time': '8',
                    'id': '2',
                    'nozzle_temp_max': '230',
                    'nozzle_temp_min': '190',
                    'remain': 14,
                    'tag_uid': '3AFCE0EF00000100',
                    'tray_color': 'FFFFFFFF',
                    'tray_diameter': '1.75',
                    'tray_id_name': 'A00-W1',
                    'tray_info_idx': 'GFA00',
                    'tray_sub_brands': 'PLA Basic',
                    'tray_type': 'PLA',
                    'tray_uuid': '952AF7737C6C426C8233E0151003B569',
                    'tray_weight': '1000',
                    'xcam_info': '34218813F401E8030000003F'},
                   {'bed_temp': '0',
                    'bed_temp_type': '0',
                    'cali_idx': -1,
                    'cols': ['F5547CFF'],
                    'ctype': 0,
                    'drying_temp': '55',
                    'drying_time': '8',
                    'id': '3',
                    'nozzle_temp_max': '230',
                    'nozzle_temp_min': '190',
                    'remain': 13,
                    'tag_uid': '5A86BDFF00000100',
                    'tray_color': 'F5547CFF',
                    'tray_diameter': '1.75',
                    'tray_id_name': 'A00-R3',
                    'tray_info_idx': 'GFA00',
                    'tray_sub_brands': 'PLA Basic',
                    'tray_type': 'PLA',
                    'tray_uuid': 'DC18908628A34CE68E572611680CCB32',
                    'tray_weight': '1000',
                    'xcam_info': '000000000000000000000000'}]}],
 'ams_exist_bits': '1',
 'ams_exist_bits_raw': '1',
 'insert_flag': True,
 'power_on_flag': True,
 'tray_exist_bits': 'f',
 'tray_is_bbl_bits': 'f',
 'tray_now': '255',
 'tray_pre': '255',
 'tray_read_done_bits': 'f',
 'tray_reading_bits': '0',
 'tray_tar': '255',
 'version': 4608}
```

AlanPeters, 2025-01-08:
Did a bit of digging. Looks like there are a number of properties that are different or missing from the AMS info in this dump. Like the 'n' value and some of the temps have different names. Was the current implementation targeting a specific printer model or AMS? Happy to contribute data or code to support the this model or version.

ohmdelta, 2025-01-08:
> Did a bit of digging. Looks like there are a number of properties that are different or missing from the AMS info in this dump. Like the 'n' value and some of the temps have different names. Was the [...]

The current ams implementation targets the p1 printers and hasn't been tested with a1 or x1