BambuTools/bambulabs_api #89: Which printers has this been tested with?

sgbaird, 2025-01-11:
In particular, for a classroom setting, I want to be able to let students control some of the print settings via Python (e.g., nozzle temperature) and print a couple lines. I'll have a separate camera for assessing print quality.

ohmdelta, 2025-01-11:
This library has been predominantly tested with P1 printers; there have been successful uses with A1 printers (see #48 ). X1 printers have yet to be tested but most of the basically functionality should work in theory.

ohmdelta, 2025-01-18:
A quick note given the recent bambulabs announcement & upcoming firmware update (see #95), it's likely that this api won't be able to support firmware 1.08 and onwards. We'll put in some documentation on this in #96 once we get a better idea of upcoming support.

closed 2025-01-18 by ohmdelta

sgbaird, 2025-01-18:
Cc @Neil-YL