BambuTools/bambulabs_api #95: Bambulabs new authentication

fidoriel, 2025-01-18:
Bambulab has announced they will introduce a new authentication scheme for “security reasons”. This will break this great software package for all printers running the new firmware.

See orca slicers issue here: https://github.com/SoftFever/OrcaSlicer/issues/8063
Their shady blog post can be found here: https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/

ohmdelta, 2025-01-18:
Yeah it's pretty disappointing and seems like a big step back for Bambu, especially given how much open source work has happened to get FDM printers to be this good (it's especially disappointing that Orcaslicer is being sidelined). Best thing we can hope for is that the community makes enough of ruckus for them to backtrack on this. 

On how this API will be affected we'll just have to see, but yes, from the looks of things we'll most likely be bricked (unless LAN mode is still a thing, which doesn't look like the case). Best thing is to keep firmware held at 1.07, and see what changes they make in 1.08 when released for P and A series.

gitagogaming, 2025-01-21:
It seems we may be safe after all?

https://blog.bambulab.com/updates-and-third-party-integration-with-bambu-connect/
[image]

ohmdelta, 2025-01-21:
Definitely an improvement but still seems like only rolling back half the change rly, from the looks of things dev mode will be entirely lan only so Bambu Handy and internet access will be accessed. Also, firmware downgrade doesn't seem to possible after 1.08 which is a downer.

JensMunkHansen, 2025-01-27:
Yes, I can confirm that a lot is broken. I can list files on my X1C and that is about it. I started a wireshark session to record what Bambulab sends to the printer. Everything is now UDP. Are anyone else looking into this?

BuSHari, 2025-04-21:
I dont know if anyone so that page, but maybe this can help:
https://wiki.bambulab.com/en/software/third-party-integration