BambuTools/bambulabs_api #98: 3D Printer Control

FranzFH, 2025-01-24:
Hi, 

so far I have tried to control my A1 using the examples. But when I use the print example it writes the file to the printer but does not start. I would like to use the printer as a 3 axis machine, is there a way that I can control the axes via GCode without a model or otherwise?

regards,

Franz

ohmdelta, 2025-01-24:
A1 hasn't been tested extensively so there might be some problem with some of the functionality - the library was originally built for P1 printers.

You can try the `Printer.gcode` endpoint (<https://mchrisgm.github.io/bambulabs_api/api/printer.html#bambulabs_api.Printer.gcode>) so send gcode to your printer - be careful with this command as the functionality can cause the nozzle to ram into the bed, etc....

FranzFH, 2025-01-27:
Thanks, it worked fine

FranzFH, 2025-01-28:
Where can i change the return values of gcode from bool to string. I need to get the Coordinates from GCode M114. M114 return should look like this: > M114
X:0.00 Y:127.00 Z:145.00 E:0.00 Count X: 0 Y:10160 Z:116000
ok

ohmdelta, 2025-02-06:
Not entirely sure this is possible, will have a look

ohmdelta, 2025-03-25:
I've had a quick look at this and I don't think it's possible with the current mqtt api. At least when publishing m114 command to mqtt there's no response.

Please let us know if you find anything on this and reopen this issue!

closed 2025-03-25 by ohmdelta