BambuTools/bambulabs_api #99: Set bed temperature issue on A1 mini

Neil-YL, 2025-01-24:
Hi. 

I have come across a weird issue when using this API on our Bambu A1 mini.

Most of the commands work as expected:

Light on/off
Pause/resume print job
Get status of printer (Job status, temperature of nozzle/bed, print speed, fan speed)
Set print speed
Capture photo from camera

However, the set_bed_temperature command doesn’t update the target temperature on the UI or change the actual bed temperature (as checked using the bed_temp command), regardless of whether a print job is in progress. Is this a limitation or a known issue?
 
Let me know if you have any suggestions or related documentation that could help. Thanks!

ohmdelta, 2025-01-24:
This library was originally written for P1 printers so A1 compatibility may vary (we don't have a A1 to test on so can't validate a lot of the functionality), so its great to hear that most of the base functionality seems to work. 

It may also be due to firmware update that setting the temperature no longer works

Neil-YL, 2025-02-14:
Hi we have been working with the API lately with our Bambu A1 mini, with fireware 1.04.00. I still cannot figure out how to set temperature via the API and I've encountered a new issue:

Previously the start print job command worked fine (I followed the example:[print_gcode.py](https://github.com/mchrisgm/bambulabs_api/blob/main/examples/print/print_gcode.py) but since this week I received a "unable parse file" error when using the same G-code file and same commands. 

[image]

I have tried all the suggested solutions from the Bambu community for similar issues, but I still can’t start the print using the API command. Printing works fine when using Bambu Studio or manually starting the job from the machine’s control panel. 

I also noticed some closed issues on this repo discussing the start print job on the A1, but I couldn’t fully grasp the context. Do you have any suggestions from previous discussion, or is there another way to start a print job via the API?

What I have tested with the API to resolve the issue:
- send file ✅
- send file and start print ❌
- manually start the print with the file sent by API with the panel on the printer ✅
- send print job from the Bambu studio and get print file name via API✅
- use last step's file name (file sent by Bambu Studio and stored on the SD card) to start print by API ❌

sgbaird, 2025-02-14:
Related: 
- https://github.com/mchrisgm/bambulabs_api/issues/98

ohmdelta, 2025-02-14:
Interesting that you were able to run the print job stuff fine before. I assume you've tied rebooting the printer and then printing? We currently only have access to p1p and p1s printers, so haven't been able to do any proper verification for a1 or x1.

My suspicion regarding the print command issue would either be a new firmware update or some underlying problem with the firmware perhaps?

Regarding the temperature stuff, you can't set the temperature with the 1.07 firmware, latest non evil firmware. Earlier firmware versions were able to set temperature with the api. I'd try rolling back your firmware to 1.02, see if that also solves the print issue

sgbaird, 2025-02-19:
@Neil-YL bump

Neil-YL, 2025-02-19:
Yes, after reset I have sent at lease two new print job to the machine via Bambu studio and tried with API as well. We will try downgrade to 1.02 tomorrow. Thanks for your information.

sgbaird, 2025-02-20:
@ohmdelta you mentioned printing with orcaslicer (https://github.com/SoftFever/OrcaSlicer, distinct from printing via Bambu Studio I'm assuming). @Neil-YL is this something you've done / could give a try? 

@ohmdelta is the reasoning here that orcaslicer uses a similar protocol for controlling the printer?

Neil-YL, 2025-02-20:
Sorry I used Bambu Studio to slice and have not tried OrcaSlicer. I would give it a try then.

ohmdelta, 2025-02-20:
Could you share a screenshot of the file structure of the unzipped 3mf file? You should be able to do this with any decompression tool. If the error is with the gcode parsing it could be an issue where the machine isn't able to find the gcode file in the archive.

sgbaird, 2025-02-20:
To clarify, this example file, correct? https://github.com/mchrisgm/bambulabs_api/blob/main/examples/print/print_3mf.py

Aside: Some of the code @Neil-YL was running is at https://github.com/AccelerationConsortium/ac-training-lab/issues/147#issuecomment-2671952659

ohmdelta, 2025-02-20:
I see. The issue is that the api when starting a print requires the path to be Metadata/plate_{YOUR_PLATE_NUMBER}.gcode. plate number by default is 1. If you follow the print gcode example it might work. It's a current limitation of the api, will fix in a bit

ohmdelta, 2025-02-21:
gcode_location = "10x10 0.1.gcode"

https://github.com/AccelerationConsortium/ac-training-lab/issues/147#issuecomment-2671952659

This line should not be modified from the example. 

You should only edit lines 12 to 13 in the example https://github.com/mchrisgm/bambulabs_api/blob/main/examples%2Fprint%2Fprint_gcode.py#L12-L13

Let us know if this works

Neil-YL, 2025-02-21:
> gcode_location = "10x10 0.1.gcode"
> 
> [AccelerationConsortium/ac-training-lab#147 (comment)](https://github.com/AccelerationConsortium/ac-training-lab/issues/147#issuecomment-2671952659)
> 
> This line should not be modified from the example.
> 
> You should only edit lines 12 to 13 in the example https://github.com/mchrisgm/bambulabs_api/blob/main/examples%2Fprint%2Fprint_gcode.py#L12-L13
> 
> Let us know if this works

Thank you so much. 
It works. I think I accidentally changed it after I want to try a new Gcode file.

closed 2025-02-28 by mchrisgm