BambuTools/bambulabs_api #101: WinError 10061 No Connection could be made because the target machine actively refused it

ndekker55, 2025-01-27:
Hi all. I am getting the above error whenever I am running the example code to grab the printer state. It then takes a second or two to"reconnect" to the machine in order to properly disconnect it. I do get the proper printer status from the machine so I am not worried there. I would like help to figure out why this is happening.  My code is below with the IP, Serial, and Access_code x'ed out. Right now I am running on windows 11 24H2 though the network to the printer. No VPN or anything special there. Thanks!

```
import time
import bambulabs_api as bl

IP = '192.168.X.XXX'
SERIAL = 'XXXXXXXXXXXX'
ACCESS_CODE = 'XXXXXXXX'

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to Bambulabs 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the Bambulabs 3D printer
    printer.connect()

    time.sleep(2)

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Turn the light off
    #printer.turn_light_off()

    #time.sleep(2)

    # Turn the light on
   # printer.turn_light_on()

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()
```

```
Starting bambulabs_api example
Connecting to Bambulabs 3D printer
IP: 192.168.X.XXX
Serial: XXXXXXXXXXXX
Access Code: XXXXXXXX
Starting camera thread.
Printer status: RUNNING
ERROR:root:Error occurred: [WinError 10061] No connection could be made because the target machine actively refused it
```

ohmdelta, 2025-01-28:
This looks like an issue with the camera thread and shouldn't be an issue for running the other apis. Are you running a x1? Camera functionality will almost definitely not work for x1 as it's a different protocol. A1 mileage may vary as well.

ohmdelta, 2025-01-28:
This looks like an issue with the camera thread; it shouldn't be an issue for running the other apis. Are you running a x1? Camera functionality will almost definitely not work for x1 as it's a different protocol. A1 mileage may vary as well.

ndekker55, 2025-01-28:
Understood. Is there a way to disable the access to the camera thread so that I can just focus on the other API callings? All I am trying to reach is the current state of the printer for a scheduling algorithm I am working on. Thanks!

ohmdelta, 2025-01-31:
not at the moment - best way is to create the mqtt & ftp clients yourself. Pr opened to address this

closed 2025-02-05 by mchrisgm

jneilliii, 2025-04-07:
Was there a regression of this issue in 2.6.0? I just installed in windows and got the same error when it attempts to start the camera thread connecting to an X1 printer. 

```
Starting bambulabs_api example
Connecting to Bambulabs 3D printer
IP: 192.168.126.9
Serial: ##
Access Code: ##
Starting camera thread.
ERROR:root:Error occurred: [WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\Users\jneil\Downloads\basic_subscription.py", line 32, in <module>
    layer_num = printer.current_layer_num()
AttributeError: 'Printer' object has no attribute 'current_layer_num'
```

my OctoPrint plugin is still able to connect and communication without issues.

jneilliii, 2025-04-07:
The above was from the subscription example, and it seems the basic example flakes out to, probably from the turn lights off command?

```
Starting bambulabs_api example
Connecting to Bambulabs 3D printer
IP: 192.168.126.9
Serial: ##
Access Code: ##
Starting camera thread.
Printer status: FINISH
ERROR:root:Error occurred: [WinError 10061] No connection could be made because the target machine actively refused it
```

Was only looking at this module as possible replacement for the pybambu module that doesn't appear to be getting updates from the upstream development project for the HA integration.

ndekker55, 2025-04-07:
This is what I did using the mqtt_client API. 
`def fetch_Bambu_printer_state(printer):
    try:
        IP = str(printer[5])
        ACCESS_CODE = str(printer[6])
        SERIAL = str(printer[7])
        printer = bl.Printer(IP, ACCESS_CODE, SERIAL)
        printer.mqtt_start()
        print_connection_attempts = 0

        while printer.mqtt_client_connected() == False and print_connection_attempts < 5:
            #print("waiting for bpm to connect to printer", flush=True)
            logging.info("waiting for Bambu-API to connect to printer")
            print_connection_attempts += 1

            time.sleep(2)

        printer_gcode_state = printer.get_state()
        #print(printer_gcode_state)

        printer_dump = printer.mqtt_dump()
        #print(printer_dump)

        time_remaining = printer.get_time()
        printer.mqtt_stop()
        return printer_gcode_state, time_remaining
        
    except Exception as e:
        logging.error(f"Failed to fetch Bambu printer state: {e}") 
        return`
It works well enough. The bambu X1C that we have returns an Error state sometimes that I am not sure what it is. I tried to figure out if it was an unknown response that the API flagged as an error but I couldn't figure it out. This gets the information I need and it has plenty left that I am not using currently.