BambuTools/bambulabs_api #113: Having trouble skipping the check for a particular GCode line

sgbaird, 2025-03-21:
`M17 R` comes from e.g., https://www.reddit.com/r/BambuLab/comments/1e84t0o/help_with_gcode/. The "R" doesn't seem to be documented in Marlin. I tried circumventing it via:

```
# restore_z_current = printer.gcode("M17 R ; restore z current")
# restore_z_current = printer.gcode("M17 X1.1 Y1.1 Z0.75 ; restore z current")  #

# HACK: bambulabs_api treating this as invalid, so circumventing check for this one
restore_z_current = printer.mqtt_client.__send_gcode_line("M17 R ; restore z current")
```
but am getting an error:
```
printer.mqtt_client.__send_gcode_line("M17 R")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: 'PrinterMQTTClient' object has no attribute '__send_gcode_line'
```

ohmdelta, 2025-03-21:
The __send_gcode_line is a private method. Please use the send_gcode function

sgbaird, 2025-03-21:
Maybe I'm misunderstanding, but this still seems to have the GCode error checking, correct?

https://github.com/mchrisgm/bambulabs_api/blob/6855b3107f2d16cde5d3e63f2eea74ae02621d2b/bambulabs_api/mqtt_client.py#L472

I'd like to temporarily disable/circumvent it for this one command, since the "R" from `M17 R` (reset motor currents from what I understand) is being recognized as invalid.

ohmdelta, 2025-03-21:
Understood - added a bypass flag

sgbaird, 2025-03-21:
Ah, thank you!

closed 2025-03-21 by ohmdelta

sgbaird, 2025-03-21:
Aside: I'll `pip install git+https://github.com/mchrisgm/bambulabs_api.git` to use the changes on the main branch

ohmdelta, 2025-03-25:
merged :)