BambuTools/bambulabs_api #118: ip and access code fail to connect

albfan, 2025-03-30:
After test with bambu studio: https://github.com/bambulab/BambuStudio/issues/5070#issuecomment-2764499283

test.py
```
#!/usr/bin/env python

import time
import bambulabs_api as bl
import sys

IP = sys.argv[1]
SERIAL = sys.argv[2]
ACCESS_CODE = sys.argv[2]

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to BambuLab 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the BambuLab 3D printer
    printer.connect()

    time.sleep(2)

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Turn the light off
    printer.turn_light_off()

    time.sleep(2)

    # Turn the light on
    printer.turn_light_on()

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()
```

```
sudo ./test.py 192.168.1.140 1234567

Connecting to BambuLab 3D printer
IP: 192.168.1.140
Serial: 1234567
Access Code: 1234567
Starting camera thread.
ERROR:root:Printer Values Not Available Yet
ERROR:root:Not connected to the MQTT server
Printer status: UNKNOWN
ERROR:root:Not connected to the MQTT server
ERROR:root:Not connected to the MQTT server
```

So same as ip and access code are not enough on bambu studio, looks it is not enough for this python module too.

anything to debug what bambu studio is doing or what this python module does to connect?

ohmdelta, 2025-03-30:
Are you using the correct serial code and access code.  Serial code should be the SN (this is the 18ish character code) number of your printer, you should be able to find this under your printer's settings menu. 

What printer, firmware versions are you using and what version of the api are you using?

albfan, 2025-03-30:
Thanks a lot, you are right. I though SN was related with PIN code or similar

```
#!/usr/bin/env python

import time
import bambulabs_api as bl
import sys

IP = sys.argv[1]
SERIAL = sys.argv[2]
ACCESS_CODE = sys.argv[3]

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to BambuLab 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the BambuLab 3D printer
    printer.connect()

    time.sleep(2)

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Turn the light off
    printer.turn_light_off()

    time.sleep(2)

    # Turn the light on
    printer.turn_light_on()

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()
```
```
./test.py 192.168.1.140 01234321C1234 98765432 #redacted
Starting bambulabs_api example
Connecting to BambuLab 3D printer
IP: 192.168.1.140
Serial: 01234321C1234
Access Code: 98765432
Starting camera thread.
Printer status: IDLE
```

and light swittch on

***BONUS*** Didn't get "Only LAN" is not exclusive, so printer can be online while you still get access to it from command line!