BambuTools/bambulabs_api #120: SSLV3_ALERT_HANDSHAKE_FAILURE on H2D

TheFiZi, 2025-04-06:
Just got my new H2D and am trying to connect to it using your python module.

When I try to connect I get the following:

```
Starting camera thread.
WARNING:root:Error in socket: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:1006)
WARNING:root:Connection failed with result code Not authorized
WARNING:root:Connection failed with result code Not authorized
ERROR:root:Printer Values Not Available Yet
ERROR:root:Not connected to the MQTT server
```

I did a quick check and all the ports are open:

```

TheFiZi, 2025-04-06:
Could it be that TLS is now enforced and your module is trying to use SSL?

Here are the results from checking what protocols/ciphers are available on my H2D:

```bash
me@localhost:~$ nmap --script ssl-enum-ciphers -p 990 my.h2d.com
Starting Nmap 7.94SVN ( https://nmap.org ) at 2025-04-05 17:34 PDT
Nmap scan report for my.h2d.com (192.168.0.99)
Host is up (0.0030s latency).

PORT    STATE SERVICE
990/tcp open  ftps
| ssl-enum-ciphers:
|   TLSv1.2:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (secp256r1) - A
|     compressors:
|       DEFLATE
|       NULL
|     cipher preference: indeterminate
|     cipher preference error: Too few ciphers supported
|_  least strength: A

Nmap done: 1 IP address (1 host up) scanned in 0.41 seconds
me@localhost:~$ nmap --script ssl-enum-ciphers -p 6000 my.h2d.com
Starting Nmap 7.94SVN ( https://nmap.org ) at 2025-04-05 17:34 PDT
Nmap scan report for my.h2d.com (192.168.0.99)
Host is up (0.0084s latency).

PORT     STATE SERVICE
6000/tcp open  X11
| ssl-enum-ciphers:
|   TLSv1.0:
|     ciphers:
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA (rsa 2048) - C
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_IDEA_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_RC4_128_MD5 (rsa 2048) - C
|       TLS_RSA_WITH_RC4_128_SHA (rsa 2048) - C
|       TLS_RSA_WITH_SEED_CBC_SHA (rsa 2048) - A
|     compressors:
|       DEFLATE
|       NULL
|     cipher preference: client
|     warnings:
|       64-bit block cipher 3DES vulnerable to SWEET32 attack
|       64-bit block cipher IDEA vulnerable to SWEET32 attack
|       Broken cipher RC4 is deprecated by RFC 7465
|       Ciphersuite uses MD5 for message integrity
|       Forward Secrecy not supported by any cipher
|   TLSv1.1:
|     ciphers:
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA (rsa 2048) - C
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_IDEA_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_RC4_128_MD5 (rsa 2048) - C
|       TLS_RSA_WITH_RC4_128_SHA (rsa 2048) - C
|       TLS_RSA_WITH_SEED_CBC_SHA (rsa 2048) - A
|     compressors:
|       DEFLATE
|       NULL
|     cipher preference: client
|     warnings:
|       64-bit block cipher 3DES vulnerable to SWEET32 attack
|       64-bit block cipher IDEA vulnerable to SWEET32 attack
|       Broken cipher RC4 is deprecated by RFC 7465
|       Ciphersuite uses MD5 for message integrity
|       Forward Secrecy not supported by any cipher
|   TLSv1.2:
|     ciphers:
|       TLS_RSA_WITH_3DES_EDE_CBC_SHA (rsa 2048) - C
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_GCM_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_GCM_SHA384 (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_IDEA_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_RC4_128_MD5 (rsa 2048) - C
|       TLS_RSA_WITH_RC4_128_SHA (rsa 2048) - C
|       TLS_RSA_WITH_SEED_CBC_SHA (rsa 2048) - A
|     compressors:
|       DEFLATE
|       NULL
|     cipher preference: client
|     warnings:
|       64-bit block cipher 3DES vulnerable to SWEET32 attack
|       64-bit block cipher IDEA vulnerable to SWEET32 attack
|       Broken cipher RC4 is deprecated by RFC 7465
|       Ciphersuite uses MD5 for message integrity
|       Forward Secrecy not supported by any cipher
|_  least strength: C

Nmap done: 1 IP address (1 host up) scanned in 1.41 seconds
me@localhost:~$ nmap --script ssl-enum-ciphers -p 8883 my.h2d.com
Starting Nmap 7.94SVN ( https://nmap.org ) at 2025-04-05 17:34 PDT
Nmap scan report for my.h2d.com (192.168.0.99)
Host is up (0.0027s latency).

PORT     STATE SERVICE
8883/tcp open  secure-mqtt
| ssl-enum-ciphers:
|   TLSv1.2:
|     ciphers:
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (secp256r1) - A
|     compressors:
|       NULL
|     cipher preference: server
|_  least strength: A

Nmap done: 1 IP address (1 host up) scanned in 0.61 seconds
```

ohmdelta, 2025-04-06:
Looks like an issue with SSL.

Can you try with this branch, set the printer version to H2D
https://github.com/BambuTools/bambulabs_api/tree/120-sslv3_alert_handshake_failure-on-h2d
This has the cert.

We don't have access to a H2D so can't confirm ourselves

TheFiZi, 2025-04-06:
Here is what I did:

```bash
me@localhost:~/venv$ python -m venv bambuapi-testing
me@localhost:~/venv$ source bambuapi-testing/bin/activate
(bambuapi-testing) me@localhost:~/temp$ git clone https://github.com/BambuTools/bambulabs_api.git
(bambuapi-testing) me@localhost:~/temp$ cd bambulabs_api/
(bambuapi-testing) me@localhost:~/temp/bambulabs_api$ git checkout 120-sslv3_alert_handshake_failure-on-h2d
(bambuapi-testing) me@localhost:~/temp/bambulabs_api$ pip install -e .

Obtaining file:///home/sa_schewee/temp/bambulabs_api
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting paho-mqtt>=2.0.0 (from bambulabs_api==2.6.0)
  Using cached paho_mqtt-2.1.0-py3-none-any.whl.metadata (23 kB)
Collecting pillow>=11.0.0 (from bambulabs_api==2.6.0)
  Using cached pillow-11.1.0-cp311-cp311-manylinux_2_28_x86_64.whl.metadata (9.1 kB)
Using cached paho_mqtt-2.1.0-py3-none-any.whl (67 kB)
Using cached pillow-11.1.0-cp311-cp311-manylinux_2_28_x86_64.whl (4.5 MB)
Building wheels for collected packages: bambulabs_api
  Building editable for bambulabs_api (pyproject.toml) ... done
  Created wheel for bambulabs_api: filename=bambulabs_api-2.6.0-0.editable-py3-none-any.whl size=5671 sha256=29a01e9713ef1a6fe4952eee7242dbb02e6557bea622efee19b6c6675a5691db
  Stored in directory: /tmp/pip-ephem-wheel-cache-bouq6hwy/wheels/e3/7d/1f/cbf6d16ab0a419e0dc5dc73ad047649bb18d4d082be802d67c
Successfully built bambulabs_api
Installing collected packages: pillow, paho-mqtt, bambulabs_api
Successfully installed bambulabs_api-2.6.0 paho-mqtt-2.1.0 pillow-11.1.0

(bambuapi-testing) me@localhost:~/apps/thegoose/gooseBambu$ python3 onMessage.py wjfgowejroif

2025-04-06 12:43:27.167410 - [gooseBambu] INFO - Connecting to printer 192.168.0.99 with serial number 0948AB520502003...
Starting camera thread.
2025-04-06 12:43:27.167945 - [gooseBambu] INFO - Waiting for printer to connect...
WARNING:root:Error in socket: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:1006)
WARNING:root:Connection failed with result code Not authorized
WARNING:root:Connection failed with result code Not authorized
WARNING:root:Connection failed with result code Not authorized
2025-04-06 12:43:32.168139 - [gooseBambu] INFO - Getting printer status...
ERROR:root:Printer Values Not Available Yet
ERROR:root:Not connected to the MQTT server
Printer Status: UNKNOWN
2025-04-06 12:43:32.168456 - [gooseBambu] INFO - Disconnecting from printer...
WARNING:root:Error in socket: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:1006)
```

The code I ran was:

```python
import re
import datetime
import time
import asyncio
import dotenv
import sys
import os
import bambulabs_api

# Load settings from .env file
dotenv.load_dotenv()
BAMBU_FQDN = os.getenv('BAMBU_FQDN')
BAMBU_SERIALNUMBER = os.getenv('BAMBU_SERIALNUMBER')
BAMBU_ACCESS_CODE = os.getenv('BAMBU_ACCESS_CODE')

# Get the current printer status with an image from the camera and reply to the user
async def getPrinterStatus(message, operatingMode):
    # Get the printer status
    printer = bambulabs_api.Printer(BAMBU_FQDN, BAMBU_SERIALNUMBER, BAMBU_ACCESS_CODE)
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Connecting to printer {BAMBU_FQDN} with serial number {BAMBU_SERIALNUMBER}...")
    printer.connect()

    # Sleep for 5 seconds not using asyncio to allow the printer to connect
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Waiting for printer to connect...")
    #await asyncio.sleep(5)
    time.sleep(5)

    # Get the printer status
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Getting printer status...")
    status = printer.get_state()

    # Format the message
    status_message = f"Printer Status: {status}"
    
    # Send the message
    if operatingMode == 1:
        print(status_message)
    else:
        await message.channel.send(status_message)

    # Disconnect from the printer
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Disconnecting from printer...")
    printer.disconnect()
```

TheFiZi, 2025-04-06:
Here is what I did:

```bash
me@localhost:~/venv$ python -m venv bambuapi-testing
me@localhost:~/venv$ source bambuapi-testing/bin/activate
(bambuapi-testing) me@localhost:~/venv$ cd ~/temp
(bambuapi-testing) me@localhost:~/temp$ git clone https://github.com/BambuTools/bambulabs_api.git
(bambuapi-testing) me@localhost:~/temp$ cd bambulabs_api/
(bambuapi-testing) me@localhost:~/temp/bambulabs_api$ git checkout 120-sslv3_alert_handshake_failure-on-h2d
(bambuapi-testing) me@localhost:~/temp/bambulabs_api$ pip install -e .

Obtaining file:///home/me/temp/bambulabs_api
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting paho-mqtt>=2.0.0 (from bambulabs_api==2.6.0)
  Using cached paho_mqtt-2.1.0-py3-none-any.whl.metadata (23 kB)
Collecting pillow>=11.0.0 (from bambulabs_api==2.6.0)
  Using cached pillow-11.1.0-cp311-cp311-manylinux_2_28_x86_64.whl.metadata (9.1 kB)
Using cached paho_mqtt-2.1.0-py3-none-any.whl (67 kB)
Using cached pillow-11.1.0-cp311-cp311-manylinux_2_28_x86_64.whl (4.5 MB)
Building wheels for collected packages: bambulabs_api
  Building editable for bambulabs_api (pyproject.toml) ... done
  Created wheel for bambulabs_api: filename=bambulabs_api-2.6.0-0.editable-py3-none-any.whl size=5671 sha256=29a01e9713ef1a6fe4952eee7242dbb02e6557bea622efee19b6c6675a5691db
  Stored in directory: /tmp/pip-ephem-wheel-cache-bouq6hwy/wheels/e3/7d/1f/cbf6d16ab0a419e0dc5dc73ad047649bb18d4d082be802d67c
Successfully built bambulabs_api
Installing collected packages: pillow, paho-mqtt, bambulabs_api
Successfully installed bambulabs_api-2.6.0 paho-mqtt-2.1.0 pillow-11.1.0

(bambuapi-testing) me@localhost:~/apps/thegoose/gooseBambu$ python3 onMessage.py wjfgowejroif

2025-04-06 12:43:27.167410 - [gooseBambu] INFO - Connecting to printer 192.168.0.99 with serial number XXXXXXXXXXX...
Starting camera thread.
2025-04-06 12:43:27.167945 - [gooseBambu] INFO - Waiting for printer to connect...
WARNING:root:Error in socket: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:1006)
WARNING:root:Connection failed with result code Not authorized
WARNING:root:Connection failed with result code Not authorized
WARNING:root:Connection failed with result code Not authorized
2025-04-06 12:43:32.168139 - [gooseBambu] INFO - Getting printer status...
ERROR:root:Printer Values Not Available Yet
ERROR:root:Not connected to the MQTT server
Printer Status: UNKNOWN
2025-04-06 12:43:32.168456 - [gooseBambu] INFO - Disconnecting from printer...
WARNING:root:Error in socket: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:1006)
```

The code I ran was:

```python
import re
import datetime
import time
import asyncio
import dotenv
import sys
import os
import bambulabs_api

# Load settings from .env file
dotenv.load_dotenv()
BAMBU_FQDN = os.getenv('BAMBU_FQDN')
BAMBU_SERIALNUMBER = os.getenv('BAMBU_SERIALNUMBER')
BAMBU_ACCESS_CODE = os.getenv('BAMBU_ACCESS_CODE')

# Get the current printer status with an image from the camera and reply to the user
async def getPrinterStatus(message, operatingMode):
    # Get the printer status
    printer = bambulabs_api.Printer(BAMBU_FQDN, BAMBU_SERIALNUMBER, BAMBU_ACCESS_CODE)
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Connecting to printer {BAMBU_FQDN} with serial number {BAMBU_SERIALNUMBER}...")
    printer.connect()

    # Sleep for 5 seconds not using asyncio to allow the printer to connect
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Waiting for printer to connect...")
    #await asyncio.sleep(5)
    time.sleep(5)

    # Get the printer status
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Getting printer status...")
    status = printer.get_state()

    # Format the message
    status_message = f"Printer Status: {status}"
    
    # Send the message
    if operatingMode == 1:
        print(status_message)
    else:
        await message.channel.send(status_message)

    # Disconnect from the printer
    print(f"{datetime.datetime.now()} - [gooseBambu] INFO - Disconnecting from printer...")
    printer.disconnect()

if __name__ == '__main__':
    # To mimic a message object from Discord for doing CLI stuff
    class message:
        content = []
    
    # Set running from CLI
    operatingMode = 1

    # Format the input into a string instead of a list
    message.content = " ".join(sys.argv[1:])

    # Verify something was input and then pass it on
    if len(message.content) == 0:
        print(f"{datetime.datetime.now()} - [gooseBambu] ERROR - Enter literally any text after the command")
    else:
        asyncio.run(getPrinterStatus(message, operatingMode))
```

ohmdelta, 2025-04-07:
Does the h2d have a developer or lan only mode? Ha bambu seems to use the cloud api for connecting to h2d atm (not sure if they can also run locally).

https://github.com/greghesp/ha-bambulab/issues/1204

TheFiZi, 2025-09-13:
Sorry for taking so long. I'm just going to close this off. I just wanted to be able to hook my Discord bot into my Bambu but not at the cost of the majority of the functionality.

closed 2025-09-13 by TheFiZi