BambuTools/bambulabs_api #123: Not working with basic config for X1C

BuSHari, 2025-04-10:
Is this api still working after the privacy changes?
Im not able to use this api with the basic config, i tried a simple mqtt client in python script and it worked.

ohmdelta, 2025-04-10:
Do you mean upgrading your firmware to 01.08 or above?

We haven't done extensive testing but earlier versions of 01.08 have had a lot of breaking changes. You might be able to use the api if developer mode is enabled on the printer

BuSHari, 2025-04-10:
I’m on above 01.08
I don’t want to switch to developer mode as I will lose the cloud connection.

I’m trying to understand, why with a simple MQTT client connection I’m able to see a response, but with this api I cannot.

ohmdelta, 2025-04-10:
Could be a couple different problems: I don't have access to an X1 but in the there have a been a couple problems with P1s

By default the api will try to publish a request for a full status update on startup. You can disable this behavior.

Are you using the basic subscription example, can you share the errors you are getting? The current example is applicable only for 2.6.1 release

Your python script, does it load in any certs? 

You might have too many existing connections to the mqtt topic or haven't closed them correctly?

BuSHari, 2025-04-11:
I just updated to the new version `2.6.1` and it seems to work now, but im getting an error on connection
```log
Starting bambulabs_api example
Connecting to BambuLab 3D printer
IP: 192.1.1.200
Serial: 00M************768
Access Code: e******f
Starting camera thread.
ERROR:root:Error occurred: [Errno 61] Connection refused
Printer status: RUNNING
```

BuSHari, 2025-04-11:
Ok i've fount it, the error happen on `camera_start()` method

ohmdelta, 2025-04-20:
@BuSHari we haven't implemented the camera integration with X1 yet (added some docs in #132). We're hoping to get something out next release - see #122 #131. Would be great if you could help with testing the X1 camera

closed 2025-04-21 by ohmdelta

BuSHari, 2025-04-21:
> [@BuSHari](https://github.com/BuSHari) we haven't implemented the camera integration with X1 yet (added some docs in [#132](https://github.com/BambuTools/bambulabs_api/pull/132)). We're hoping to [...]

Yes i can help with testing...