BambuTools/bambulabs_api #157: Swapped commands load and unload filament

Drogon-code, 2025-08-01:
Hi, basic info:
Printer: A1 mini
Firmware: 01.05.00.00

I run command printer.unload_filament and my printer start load filament. If i run commnad load_filament the printer starts unload filemant. I have only printer A1 mini and I can't tested this commands on difrent printers. Can you please look into this?

Thank you

PS: Can I send command load filament for specific AMS tray?

Zynth9999, 2025-09-16:
how did you get it to load/unload? i cant get it to do anything like that on 01.05.xx on a1 mini

ohmdelta, 2025-09-16:
Are you able try older firmware versions

ohmdelta, 2025-09-16:
would you be able to confirm if the unload & load commands are reversed?

Zynth9999, 2025-09-17:
For me the commands are working correctly, unload unloads and load loads.

ohmdelta, 2025-09-17:
Thanks, we'll add some documentation for this

closed 2025-09-17 by ohmdelta