BambuTools/bambulabs_api #162: A1 not getting status

NelsonLive, 2025-08-26:
Using the basic code on my A1 and getting STATUS UKNOWN

IP = '192.168.1.200' fake
SERIAL = '02919C430100335' fake
ACCESS_CODE = '10892895' fake

[image]

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to BambuLab 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

   # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the Bambulabs 3D printer
    printer.connect()

    time.sleep(2)

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    # Turn the light off
    printer.turn_light_off()

    time.sleep(2)

    # Turn the light on
    printer.turn_light_on()

    # Disconnect from the Bambulabs 3D printer
    printer.disconnect()

NelsonLive, 2025-08-26:
[image]

I get this on the loop. It works the second time but what I need to do then?

NelsonLive, 2025-08-26:
Thank you very much adam, I can now automate my 3d print farm with amazon :)

NelsonLive, 2025-08-27:
Yes, that is what I have done and it doesnt start the print.

NelsonLive, 2025-08-27:
> What version firmware are you on? If 1.06 I think you need to go to LAN,
> Turn on Lan Mode Only, then turn on Developer Mode

Requirement already satisfied: bambulabs_api in
c:\users\nel\appdata\local\programs\python\python313\lib\site-packages
(2.6.3)

2.6.3 is the version I think.

Missatge de adamw005 ***@***.***> del dia dc., 27 d’ag. 2025
a les 18:52:

NelsonLive, 2025-08-27:
> the firmware of your printer

01.05.00.00

Missatge de adamw005 ***@***.***> del dia dc., 27 d’ag. 2025
a les 19:08:

ohmdelta, 2025-08-27:
Have you provided the correct path to the file on the printer? Saving the file via bambu studio/orcaslicer places the file in the cache folder by default

NelsonLive, 2025-08-27:
I tried this, is the correct way? I sent the file as you said.

printer.start_print("/cache/CUBATA.3mf", plate_number=1, use_ams=True,
ams_mapping=[0])

Missatge de ohmdelta ***@***.***> del dia dc., 27 d’ag. 2025
a les 21:27:

NelsonLive, 2025-08-28:
I have put in developer mode and this is my code right now, shows like the response is True but it doesn't start any print:

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to BambuLab 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

   # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the Bambulabs 3D printer
    printer.connect()

    time.sleep(2)

status = printer.get_state()
print(f'Printer status: {status}')

print(printer.start_print("/cache/CUBATA.gcode.3mf", plate_number=1, use_ams=True, ams_mapping=[0]))

And this is the response I get:

Starting camera thread.
Printer status: IDLE
True

NelsonLive, 2025-08-29:
> I would pop the SD card in to your pc and make sure the file path is
> correct

I have done that and happens the same, shows true and doesn't start any
print.

Is there a way to debug whats happening?

Missatge de adamw005 ***@***.***> del dia dj., 28 d’ag. 2025
a les 15:37: