BambuTools/bambulabs_api #165: Can not pause print job

GeneArnold, 2025-09-07:
I have the basic example working fine. I can toggle the light and get printer status. I was hoping to pause a print job but the pause call runs but has no effect. I know that bambu has locked a lot of things down now. Was this one of the functions?

GeneArnold, 2025-09-19:
I wanted to follow up on this. Is there more information I can offer to help figure this out? Should I switch over to LAN only mode? I'm a bit new to this and I apologize if these are items I should already be aware of. Thank you!

abwebc, 2026-09-18:
Very likely the same root cause as #183: `pause_print()` publishes `{"print": {"command": "pause"}}` with no `sequence_id` and no `param`, and the firmware discards such a message without any reply — which is exactly "runs but has no effect". Adding `"sequence_id"` and `"param": ""` made `resume` work instantly on a P2S here, with a `result: SUCCESS` ack on the report topic. Not locked down by Bambu, just a malformed payload.